YosysHQ/yosys #5215: log_const thinks a ring buffer and prayer add up to an allocator

widlarizer, 2025-07-08:
### Version

66035f70

### On which OS did this happen?

Linux

### Reproduction Steps

Related to #5210. In the words of an artist, "trust me, it gets so much worse". Dump this in your pass of choice:

```cpp
std::vector<RTLIL::Const> consts;
for (int i = 0; i < 110; i++) {
	consts.push_back(Const(i));
}
std::vector<const char*> const_logs;
for (auto c : consts) {
	const_logs.push_back(log_const(c));
}
for (auto l : const_logs) {
	std::cout << l << "\n";
}
```

### Expected Behavior

Numbers from 0 to 110 get printed. `log_const` actively is built to allow returned strings to live longer than the next call to `log_const`. At least give me an error if some magic hidden resource runs out

### Actual Behavior

This won't print numbers 0..8 because `log_const` uses an absurd ring buffer for leaking temporaries and just *hopes* that you never keep a pointer to a const after you've logged 100 more. Instead you get this because newer logs overwrite the ring buffer entries:
```
101
102
[...]
108
109
9
10
11
12
[...]
108
109
```

Ideally we just return `std::string` instead of this ring buffer. Maybe finally an excuse to deprecate `log_*` functions that do string formatting instead of logging - tag them deprecated, replace their use with `std::string` functions in the codebase.

Also the same thing exists in nextpnr as the `StrRingBuffer`. I assume this bug isn't discoverable in practice in either program currently but it's a classic foot gun

KrystalDelusion, 2025-07-08:
10/10 issue name

rocallahan, 2025-07-11:
LOL. Yes we need to get rid of `string_buf` too.