YosysHQ/yosys #5220: Wrapped operator synthesis gives unhandled state 1'z error

precisionmoon, 2025-07-10:
### Version

Yosys 0.55 (git sha1 60f126cd0, g++ 13.3.0-6ubuntu2~24.04 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

Run ORFS with SYNTH_WRAPPED_SYNTHESIS=1 for the following design:

sky130hd/microwatt

### Expected Behavior

No failure

### Actual Behavior

22.16. Executing PORTARCS pass. (derive propagation arcs)
22.17. Executing ABC9_OPS pass (helper functions for ABC9).
22.18. Mapping module 'bridge'.
22.18.1. Executing ABC9_OPS pass (helper functions for ABC9).
22.18.2. Executing XAIGER2 backend.
22.18.3. Executing ABC9_EXE pass (technology mapping using ABC9).
22.18.4. Executing ABC9.
22.18.5. Executing XAIGER2 frontend.
22.19. Mapping module 'cache_ram_5_64_1489f923c4dca729178b3e3233458550d8dddf29'.
22.19.1. Executing ABC9_OPS pass (helper functions for ABC9).
22.19.2. Executing XAIGER2 backend.
22.19.3. Executing ABC9_EXE pass (technology mapping using ABC9).
22.19.4. Executing ABC9.
22.19.5. Executing XAIGER2 frontend.
22.20. Mapping module 'control_3_bf8b4530d8d246dd74ac53a13471bba17941dff7'.
22.20.1. Executing ABC9_OPS pass (helper functions for ABC9).
22.20.2. Executing XAIGER2 backend.
22.20.3. Executing ABC9_EXE pass (technology mapping using ABC9).
22.20.4. Executing ABC9.
22.20.5. Executing XAIGER2 frontend.
22.21. Mapping module 'core_debug_0'.
22.21.1. Executing ABC9_OPS pass (helper functions for ABC9).
22.21.2. Executing XAIGER2 backend.
22.21.3. Executing ABC9_EXE pass (technology mapping using ABC9).
22.21.4. Executing ABC9.
22.21.5. Executing XAIGER2 frontend.
22.22. Mapping module 'cr_file_0_5ba93c9db0cff93f52b521d7420e43f6eda2784f'.
22.22.1. Executing ABC9_OPS pass (helper functions for ABC9).
22.22.2. Executing XAIGER2 backend.
ERROR: Unhandled state 1'z
Command exited with non-zero status 1

KrystalDelusion, 2025-07-11:
This is the Yosys repository, can you reduce this to the Yosys component only please?

precisionmoon, 2025-07-11:
I re-created a Yosys stand-alone testcase.   This can be accessed from here.
https://drive.google.com/file/d/1Z42k7WfDA9Qug8-Ny1CpeVIxxvMr7MJ_/view?usp=drive_link

KrystalDelusion, 2025-07-16:
Ping @povik

precisionmoon, 2025-07-19:
@povik, can you advise on the next steps?   We do see stdundef -zero in synth.tcl.   How do we resolve this issue?

povik, 2025-07-21:
I think `setundef` replaces X, not Z. One sure way to resolve this is to modify `abc_new`/`write_xaiger2` so that it behaves the same as `abc`.

@precisionmoon I'm not sure there's a straightforward fix we can implement by changing the script only.

KrystalDelusion, 2025-07-21:
> I think `setundef` replaces X, not Z

I couldn't find anything in the code, but when I tested with this example and it did resolve it to 13'h0

closed 2025-07-21 by KrystalDelusion

precisionmoon, 2025-07-28:
I tried to move "stdundef -zero" before ABC optimization in ORFS synth.tcl.   This changed QoR for many designs.
4 designs show worse QoR:
asap7/aes-block: 530% worse sTNS
asap7/mock-array: 419% worse sTNS
nangate45/ariane133: 100% worse sTNS
sky130hd/microwatt: 7.6% worse sTNS

6 designs show better QoR:
asap7/cva6: 100% better sTNS
nangate45/bp_fe: 53% better sTNS
nangate45/swerve_wrapper: 43% better sTNS
nangate45/bp_be:  31.5% better sTNS
nangate45/tinyRocket: 19% better sTNS
nangate45/bp_multi: 3% better sTNS