YosysHQ/yosys #5223: Bufnorm crashes on bi-directional drivers in design

povik, 2025-07-11:
### Version

Yosys 0.55+23 (git sha1 e57a2b944, clang++ 11.1.0 -fPIC -O3)

### On which OS did this happen?

macOS

### Reproduction Steps

```
read_verilog <<EOF
(*blackbox*)
module foo(inout x);
endmodule
module top();
	wire w;
	foo a(.x(w));
	foo b(.x(w));
endmodule
EOF
bufnorm
```

### Expected Behavior

No crash and valid "bufnormed" representation

### Actual Behavior

```
2. Executing BUFNORM pass (convert to buffer-normalized form).
Buffer-normalizing module top.
  directly driven by cell b port x: w
  directly driven by cell a port x: w
Summary: removed 0, updated 0, kept 0, and created 0 buffers, and updated 0 cell ports.
ERROR: Assert `wire->driverCell_ == nullptr' failed in kernel/rtlil.cc:4061.
```

povik, 2025-07-11:
The same crash happens in `abc_new` on a design with multiple bi drivers (in the form of technology primitives) on a net.

KrystalDelusion, 2025-07-16:
I'm not very familiar with buffer normalised form, but what is a "valid buformed representation" for a cell with multiple drivers?  Or is it just the wires that can only have a single driver?

jix, 2025-08-08:
I've started looking into this. Beyond this specific crash there are a lot issues with bufnorm and bi-directional connections / inout ports. There seems to be no handling of inout module or cell ports anywhere in the current bufnorm code. All places that are concerned with port direction either interpret those ports as pure input or pure output ports in a way that feels arbitrary to me. This doesn't only cause crashes, I also found instances where bufnorm silently loses bidirectional connections.

For adding bi-directional connection / inout port support, I think it makes sense to distinguish between a) the bufnorm handling in the RTLIL kernel, which actively maintains the bufnorm invariants as the design is modified, but comes with some requirements on the design when entered, and b) the `bufnorm` pass, which brings arbitrary RTLIL into a form where the kernel functionality can deal with it.

My current plan is to
* make sure that the RTLIL kernel doesn't crash in the presence of inout ports
* make sure the kernel preserves inout connectivity, i.e. we don't get any semantic changes when entering and leaving bufnorm mode, without the need for any wire <-> wire connections
* make sure the kernel maintains the `driverCell` pointers also for inout ports that are directionally unambiguous (i.e. they're connected to a driver or not connected to any other potential driver)
* implement a new `bufnorm` pass using a `DriverMap` to index the port <-> port connectivity. The current `bufnorm` uses a `SigMap` that only indexes wire connectivity and that is insufficient to correctly handle inout ports in general (unless augmented with an ad-hoc reimplementation of what `DriverMap` provides).

On the kernel side I've done enough prototyping that I don't expect any surprises here as long as we're free to use some form of bidirectional buffer cell to represent directionally ambiguous connections (whether that's a `$buf` with an attribute, an opt barrier after we get those or something new). I think it should be doable even without extra cells, but it's cumbersome to properly test that without an inout-aware `bufnorm` pass, so I want to come back to this later. Also, if there's a use case for being able to modify parts of the design that have directionally ambiguous connections while in bufnorm mode, it may actually be advantageous to have those be represented as explicit cells.

For the `bufnorm` pass there's also the open question of how much of its current behavior to replicate, e.g. which existing wires are used and how buffers are chained. It also has a several options that affect this and I'm not sure whether there's really a need for all that flexibility or whether that's just left over from the initial prototyping.

jix, 2025-09-16:
This RTLIL-kernel side (that produces the error) of this is fixed by #5353. That PR is the result of further iterating on the design I sketched in my previous comment. For more details see the PR.

While this fixes the reproducer and the underlying kernel side problem, there are still issues with bi-directional connections in the experimental `bufnorm` utility pass itself, which #5353 doesn't fix. Until those are also fixed, only `bufnorm -update` or the equivalent kernel API (currently used by all passes that implicitly use bufnorm) should be used to enter buffer normalized mode.