YosysHQ/yosys #5225: Signed RHS is not sign-extended in part-select assignment (IEEE 1364-2005 §5.5.3 violation)

1353369570, 2025-07-12:
### Version

Yosys 0.53 

### On which OS did this happen?

Linux

### Reproduction Steps

In Verilog 2005, a signed constant assigned to a wider left-hand side (e.g., via part-select) should be sign-extended. However, Yosys fails to do so when a signed constant like 3'sb111 is assigned to a 4-bit part-select. Instead, it treats the RHS as unsigned and zero-extends it, resulting in incorrect synthesis behavior.

This violates IEEE Std 1364-2005, Section 5.5.3, which explicitly defines that:

“If needed, extend the size of the right-hand side, performing sign extension if, and only if, the type of the right-hand side is signed.”

Minimal Example：
module test_sign_extend (
    input  wire [2:0] x,
    output reg  [15:0] y_ref,
    output reg  [15:0] y
);
always @(*) begin
    y_ref = 16'd0;
    y     = 16'd0;

    // Expected: y_ref[x+:4] == 4'b1111
    y_ref[x +: 4] = 4'hf;

    // Bug: y[x+:4] gets 4'b0111 instead of 4'b1111 (sign extension missing)
    y[x +: 4] = 3'sb111;
end
endmodule

read_verilog test.v
proc; clean -purge; write_verilog -noattr

### Expected Behavior

3'sb111 should be sign-extended to 4'b1111 before being assigned to the 4-bit slice.

### Actual Behavior

But Yosys produces:

assign y = $signed(_3_) < 0 ? 4'hf << -_3_ : 4'hf >> _3_;
This means the signedness is ignored and the result is incorrect.

KrystalDelusion, 2025-07-14:
Seems like it isn't correctly handing the signedness of the RHS; the RTLIL shows that `A_SIGNED` isn't set on the `$shift` cell:
```
cell $shift $shift$test.v:0$20
  parameter \A_SIGNED 0
  parameter \A_WIDTH 4
  parameter \B_SIGNED 1
  parameter \B_WIDTH 33
  parameter \Y_WIDTH 16
  connect \A 4'1111
  connect \B $neg$test.v:0$19_Y
  connect \Y $shift$test.v:0$20_Y
end
```
Changing that by hand gives the following verilog output:
```
assign y = $signed(_3_) < 0 ? $signed(4'hf) << - _3_ : $signed(4'hf) >> _3_;
```

KrystalDelusion, 2025-07-14:
Also your description is wrong

> 3'sb111 should be sign-extended to 4'b1111 before being assigned to the 4-bit slice.

The code *does* use `4'hf`, which is equivalent to `4'b1111`, not `4'b0111`.  What is the problem here?

closed 2025-07-14 by KrystalDelusion