YosysHQ/yosys #5242: Mishandling of characters in VHDL parameters

KrystalDelusion, 2025-07-22:
### Version

Yosys 0.55+23 (git sha1 e57a2b944, ccache clang++ 18.1.8 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

string.vhd
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity top is
  port (
    a : out character
  );
end entity;

architecture rtl of top is
component sub is
  generic (
    s : character
  );
  port (
    a : out character
  );
end component;
begin
  sub_i: sub
    generic map (
        s => '1'
    )
    port map (
        a => a
    );
end architecture;
```
sub.vhd
```vhdl
entity sub is
  generic (
    s : character := 'a'
  );
  port (
    a : out character
  );
end entity;

architecture rtl of sub is
begin
  a <= s;
end architecture;
```
Run `yosys string.vhd sub.vhd -p 'hierarchy -top top'`

### Expected Behavior

Yosys completes normally, as it does when there is no generic and `a <= '1';` directly.

### Actual Behavior

```
VERIFIC-INFO [VHDL-1067] string.vhd:5: processing 'top(rtl)'
VERIFIC-INFO [VHDL-1067] sub.vhd:1: processing '\sub(s='1')\(rtl)'
Importing module top.
Importing module \\sub(s='1')\(rtl).
ERROR: Error parsing VHDL character.
```

This appears to be coming from the following function: https://github.com/YosysHQ/yosys/blob/2223d7848be5a249b885c355904aa1142081ea23/frontends/verific/verific.cc#L273-L278

It seems that character extraction assumes that characters start and end with double quotes.  Most characters start and end with single quotes rather than double quotes, but some don't have quotes at all.  They're also only ever 8bits rather than 32.  Having a wire of type `character`, we also see this in the resulting RTLIL:
```
  attribute \enum_value_11111111 "\\'\377'"
  attribute \enum_value_11111110 "\\'\376'"
...
  attribute \enum_value_10100000 "\\'\240'"
  attribute \enum_value_10011111 "\\c159"
  attribute \enum_value_10011110 "\\c158"
...
  attribute \enum_value_10000000 "\\c128"
  attribute \enum_value_01111111 "\\del"
  attribute \enum_value_01111110 "\\'~'"
  attribute \enum_value_01111101 "\\'}'"
  attribute \enum_value_01111100 "\\'|'"
  attribute \enum_value_01111011 "\\'{'"
  attribute \enum_value_01111010 "\\'z'"
  attribute \enum_value_01111001 "\\'y'"
...
  attribute \enum_value_01100001 "\\'a'"
  attribute \enum_value_01100000 "\\'`'"
  attribute \enum_value_01011111 "\\'_'"
  attribute \enum_value_01011110 "\\'^'"
  attribute \enum_value_01011101 "\\']'"
  attribute \enum_value_01011100 "\\'\\'"
  attribute \enum_value_01011011 "\\'['"
  attribute \enum_value_01011010 "\\'Z'"
  attribute \enum_value_01011001 "\\'Y'"
...
  attribute \enum_value_01000001 "\\'A'"
  attribute \enum_value_01000000 "\\'@'"
  attribute \enum_value_00111111 "\\'?'"
  attribute \enum_value_00111110 "\\'>'"
  attribute \enum_value_00111101 "\\'='"
  attribute \enum_value_00111100 "\\'<'"
  attribute \enum_value_00111011 "\\';'"
  attribute \enum_value_00111010 "\\':'"
  attribute \enum_value_00111001 "\\'9'"
  attribute \enum_value_00111000 "\\'8'"
  attribute \enum_value_00110111 "\\'7'"
  attribute \enum_value_00110110 "\\'6'"
  attribute \enum_value_00110101 "\\'5'"
  attribute \enum_value_00110100 "\\'4'"
  attribute \enum_value_00110011 "\\'3'"
  attribute \enum_value_00110010 "\\'2'"
  attribute \enum_value_00110001 "\\'1'"
  attribute \enum_value_00110000 "\\'0'"
  attribute \enum_value_00101111 "\\'/'"
  attribute \enum_value_00101110 "\\'.'"
  attribute \enum_value_00101101 "\\'-'"
  attribute \enum_value_00101100 "\\','"
  attribute \enum_value_00101011 "\\'+'"
  attribute \enum_value_00101010 "\\'*'"
  attribute \enum_value_00101001 "\\')'"
  attribute \enum_value_00101000 "\\'('"
  attribute \enum_value_00100111 "\\'''"
  attribute \enum_value_00100110 "\\'&'"
  attribute \enum_value_00100101 "\\'%'"
  attribute \enum_value_00100100 "\\'$'"
  attribute \enum_value_00100011 "\\'#'"
  attribute \enum_value_00100010 "\\'\"'"
  attribute \enum_value_00100001 "\\'!'"
  attribute \enum_value_00100000 "\\' '"
  attribute \enum_value_00011111 "\\usp"
  attribute \enum_value_00011110 "\\rsp"
  attribute \enum_value_00011101 "\\gsp"
  attribute \enum_value_00011100 "\\fsp"
  attribute \enum_value_00011011 "\\esc"
  attribute \enum_value_00011010 "\\sub"
  attribute \enum_value_00011001 "\\em"
  attribute \enum_value_00011000 "\\can"
  attribute \enum_value_00010111 "\\etb"
  attribute \enum_value_00010110 "\\syn"
  attribute \enum_value_00010101 "\\nak"
  attribute \enum_value_00010100 "\\dc4"
  attribute \enum_value_00010011 "\\dc3"
  attribute \enum_value_00010010 "\\dc2"
  attribute \enum_value_00010001 "\\dc1"
  attribute \enum_value_00010000 "\\dle"
  attribute \enum_value_00001111 "\\si"
  attribute \enum_value_00001110 "\\so"
  attribute \enum_value_00001101 "\\cr"
  attribute \enum_value_00001100 "\\ff"
  attribute \enum_value_00001011 "\\vt"
  attribute \enum_value_00001010 "\\lf"
  attribute \enum_value_00001001 "\\ht"
  attribute \enum_value_00001000 "\\bs"
  attribute \enum_value_00000111 "\\bel"
  attribute \enum_value_00000110 "\\ack"
  attribute \enum_value_00000101 "\\enq"
  attribute \enum_value_00000100 "\\eot"
  attribute \enum_value_00000011 "\\etx"
  attribute \enum_value_00000010 "\\stx"
  attribute \enum_value_00000001 "\\soh"
  attribute \enum_value_00000000 "\\nul"
  attribute \wiretype "\\character"
  attribute \src "string.vhd:7.5-7.6"
  wire width 8 output 1 \a
```

which implies that Yosys is already getting a map of 8bit values to their character representation.  Maybe we can use that somehow?

nakengelhardt, 2025-07-22:
Admittedly I barely know any VHDL, but that seems like a straight error, AFAIK it's mandatory to use single quotes for character literals, and double quotes are only for strings (same as single-bit constant vs vector constant)?

KrystalDelusion, 2025-07-22:
Yep, changing the code to use double quotes for a character causes Verific to reject it.  I didn't test attributes, but this function also gets used for those, and it might be possible to use a single character in attributes without Verific rejecting it (though at that point it's a single width string, so I'm not sure if it would use the character parsing or the string parsing).