YosysHQ/yosys #5274: Regression: Assert `count_id(wire->name) == 0' failed in ../yosys-src/kernel/rtlil.cc:2621 when building Glasgow Interface Explorer applets in 0.56

joelpmichael, 2025-08-08:
### Version

Yosys 0.56 (git sha1 9c447ad9d, ccache clang++ 18.1.3 -O3 -flto -flto)

### On which OS did this happen?

macOS

### Reproduction Steps

As requested by @whitequark this appears to be an issue with Yosys. Tracking as https://github.com/GlasgowEmbedded/glasgow/issues/994

1. Follow Glasgow Interface Explorer installation as per https://glasgow-embedded.org/latest/install.html at commit 8b02130
2. `glasgow -v run selftest` (or any other applet)

### Expected Behavior

Glasgow applet compiles and runs successfully.

### Actual Behavior

Build fails with final error:

```
T: glasgow.hardware.build_plan: build: ERROR: Assert `count_id(wire->name) == 0' failed in ../yosys-src/kernel/rtlil.cc:2621.
E: glasgow.applet.internal.selftest: gateware build failed with exit code 1; see build log above for details
```

Full compile log:
[glasgow-run-selftest.log](https://github.com/user-attachments/files/21681257/glasgow-run-selftest.log)

I'm not very familiar with yosys internals, so any help with narrowing down where this issue is would be appreciated!

georgerennie, 2025-08-08:
This is happening in the AIGER frontend within synth_ice40 so its probably the same as #4451 / #4349 again. It looks like @Ravenslofty had proposed adding splitnets to fix it but this didn't actually happen in the end?

whitequark, 2025-08-08:
It looks from th linked issues that a revert may not be feasible and it has to be dealt with properly. The proposed `splitnets` looks like a huge hack to me.

whitequark, 2025-08-08:
> I don't see any commits since 0.55 that seem that directly related on first glance?

I started bisecting and I was wrong earlier: 0.55 is also bad. Since 0.54 and 0.55 weren't tested with Glasgow due to #5182 that probably slipped in too...

whitequark, 2025-08-08:
> Yeah I'm not sure, also tried a couple builds I had lying around from the past few months (some with pyosys/sanitizers, some not) and none reproed. I was able to repro it with `yowasp-yosys`.

I can repro it with native debug builds on amd64 going as far back as 0.52.

KrystalDelusion, 2025-08-14:
I just tested this on debian Yosys 0.56+105 (git sha1 195d3ef94, ccache g++ 12.2.0-14+deb12u1 -fPIC -O3) and
Yosys 0.52 (git sha1 fee39a328, ccache clang++ 14.0.6 -fPIC -O3), ubuntu 24.04 Yosys 0.56+105 (git sha1 195d3ef94, ccache clang++ 18.1.3 -fPIC -O3) and Yosys 0.55+3 (git sha1 478b6a2b3, ccache clang++ 18.1.3 -fPIC -O3) and can't reproduce this on any of them.  Currently building `yowasp-yosys` to test with that, but it's taking a while.

I did however discover a bug with `rename -unescape` while looking into this, so that's not (currently) a viable workaround here.

KrystalDelusion, 2025-08-15:
With `yosys`:
```
2.41.12.6. Executing AIGER frontend.
#X# New IdString '\top.fx2_crossbar.bus.fd$abc9' with index 1332.
M=8 I=8 L=0 O=8 A=0
#X# New IdString '$i1' with index 1330.
#X# New IdString '$aiger575$1' with index 1487.
#X# Connect (SigSig) in top.fx2_crossbar.bus.fd$abc9: $aiger575$1 = $i1 (1 bits)
#X# New IdString '$i2' with index 1335.
#X# New IdString '$aiger575$2' with index 1486.
#X# Connect (SigSig) in top.fx2_crossbar.bus.fd$abc9: $aiger575$2 = $i2 (1 bits)
#X# New IdString '$i3' with index 1350.
#X# New IdString '$aiger575$3' with index 1389.
#X# Connect (SigSig) in top.fx2_crossbar.bus.fd$abc9: $aiger575$3 = $i3 (1 bits)
...
```

With `yowasp-yosys`:
```
2.41.12.6. Executing AIGER frontend.
#X# New IdString '\top.fx2_crossbar.bus.fd$abc9' with index 1355.
M=8 I=8 L=0 O=8 A=0
#X# New IdString '$i9' with index 1488.
#X# New IdString '$aiger575$1' with index 1336.
#X# Connect (SigSig) in top.fx2_crossbar.bus.fd$abc9: $aiger575$1 = $i9 (1 bits)
#X# New IdString '$i6' with index 1344.
#X# New IdString '$aiger575$2' with index 1342.
#X# Connect (SigSig) in top.fx2_crossbar.bus.fd$abc9: $aiger575$2 = $i6 (1 bits)
ERROR: Assert `count_id(wire->name) == 0' failed in ../yosys-src/kernel/rtlil.cc:2617.
```

Why are they different? And what is even happening here https://github.com/YosysHQ/yosys/blob/bf625951d774d2747fa606f923ba5cf34b935a4a/frontends/aiger/aigerparse.cc#L659

KrystalDelusion, 2025-08-15:
Ping @rocallahan is it possible this `stringf` broke with #5221 ?

rocallahan, 2025-08-15:
> Ping @rocallahan is it possible this stringf broke with https://github.com/YosysHQ/yosys/pull/5221 ? Because $i9 shouldn't be possible with I=8, unless there is something weird with how stringf("$i%0*d", digits, i) is being parsed.

Anything's possible, but that `top.il` testcase passes for me and it's also clean with undefined and address sanitizers. Also a unit test of `stringf("$i%0*d", 1, 8)` produces `$i8` as expected, and is also clean with undefined and address sanitizers.

KrystalDelusion, 2025-08-15:
It might be something specific to how strings get handled in the WASI build, I'm not sure exactly what changes, but I can only get the error to reproduce under WASI.

rocallahan, 2025-08-15:
I can reproduce this under WASI. It's definitely my fault. My code assumes that calling a varargs function with a `int` parameter where the function is expecting an `int` parameter is fine and will just silently truncate the `long long` to `int`. This is generally true for C where the parameters are normally passed in registers, but it's not true for WebAssembly:
https://github.com/WebAssembly/tool-conventions/blob/main/BasicCABI.md
> Varargs are placed in a buffer by the caller and the last parameter to the function is a pointer to that buffer. The callee is allowed to modify the contents of the buffer, so the caller is responsible for making a copy of any varargs data that the callee should not modify. Arguments are arranged in order in the buffer, with each argument placed at the lowest appropriately aligned address after the previous argument.

So given the format `%0*d`, `vsprintf` is expecting two `int` parameters at offsets 0 and 4 in that buffer, but we're passing `int` and `long long` and putting them at offsets 0 and 8 in that buffer. `vsprintf` reads the second `int` parameter from uninitialized memory.

rocallahan, 2025-08-15:
PR #5292 should fix this. WFM locally.

closed 2025-08-15 by KrystalDelusion