YosysHQ/yosys #5279: Yosys 0.56 timeout

eder-matheus, 2025-08-11:
### Version

Yosys 0.56 (git sha1 9c447ad9d, g++ 13.3.0-6ubuntu2~24.04 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

[ariane133_issue.tar.gz](https://github.com/user-attachments/files/21723923/ariane133_issue.tar.gz)

`tar -xf ariane133_issue.tar.gz`
`cd ariane133_issue`
`./run-me-ariane133-nangate45-base.sh`

### Expected Behavior

Yosys run ends with no errors in a reasonable time.

### Actual Behavior

After updating Yosys to the 0.56 version in OpenROAD-flow-scripts, two designs are failing with a timeout during 

```
Yosys execution. All of them are stuck here:
[...]
[2025-08-07T15:03:30.126Z] 3.7.8. Executing FSM_MAP pass (mapping FSMs to basic logic).
[2025-08-07T15:03:30.126Z] 3.8. Executing OPT pass (performing simple optimizations).
[2025-08-07T15:03:30.126Z] 3.8.1. Executing OPT_EXPR pass (perform const folding).
[2025-08-07T15:03:30.126Z] 3.8.2. Executing OPT_MERGE pass (detect identical cells).
[2025-08-07T15:03:30.126Z] 3.8.3. Executing OPT_MUXTREE pass (detect dead branches in mux trees).
[2025-08-07T15:03:32.767Z] 3.8.4. Executing OPT_REDUCE pass (consolidate $*mux and $reduce_* inputs).
[2025-08-07T15:03:32.767Z] 3.8.5. Executing OPT_MERGE pass (detect identical cells).
[2025-08-07T15:03:32.767Z] 3.8.6. Executing OPT_DFF pass (perform DFF optimizations).
```

The designs are nangate45/ariane133 and nangate45/ariane136. The attached test case reproduces the error for nangate45/ariane133.

Let me know if I need to provide any extra information or a different format for the test case.

Anhijkt, 2025-08-12:
https://github.com/YosysHQ/yosys/pull/5287 seems to make execution time of `5279.il` test adequate. Seems like my PR was really under-tested. While working on it I used [picorv32](https://github.com/YosysHQ/picorv32) design as a test. 
Running `synth -top picorv32` on pre-https://github.com/YosysHQ/yosys/pull/5241 yosys gives the following stats -
```
   Number of wires:               6223
   Number of wire bits:           8796
   Number of public wires:         180
   Number of public wire bits:    2305
   Number of ports:                 27
   Number of port bits:            409
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               8038
   ...
```

While post-https://github.com/YosysHQ/yosys/pull/5241 gives -
```
   Number of wires:               6212
   Number of wire bits:           8785
   Number of public wires:         180
   Number of public wire bits:    2305
   Number of ports:                 27
   Number of port bits:            409
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               8027
   ...
```
I probably should've include this metrics in original PR.

eder-matheus, 2025-09-04:
Yosys 0.57 doesn't timeout anymore. Thanks for the fix!

closed 2025-09-04 by eder-matheus