YosysHQ/yosys #5296: Module-level parallelism

akashlevy, 2025-08-17:
### Feature Description

For very large designs, passes can take a very long time. I'm wondering if anyone has tried implementing module-level parallelism for any of the passes, especially `opt` (and its subcomponents) and `abc`.

rocallahan, 2025-11-13:
Can you publish a branch so we can see what you did?

akashlevy, 2025-11-14:
I don't have time to extricate the changes from the rest, but you can find them here: https://github.com/Silimate/yosys/blob/main/passes/techmap/abc.cc

phsauter, 2025-12-17:
> This is not easy to do because RTLIL is generally not compatible with multithreading

Agreed, something I did previously is to go to flatten as fast as possible where the hierarchy I want to keep anyway is kept, then writeout the netlist, spawn more Yosys' in separate threads (via `xargs`) and then read in the netlist and let each process its own module.  
This worked reasonably well but was a nightmare to maintain so I eventually gave up.

Anyway, my point is that maybe something similar could be done instead of making RTLIL multithreading ready. Meaning in a yosys script there would be some `fork` and `join` commands and when Yosys encounters a `fork`, it writes out all current modules, starts itself in new threads, lets them work through the script until the next `join` at which point it reads the result back in module for module and continues single threaded.  
This would still require work, most of it to make sure a somewhat efficient and good solution exists to save and reload RTLIL exists but it might be easier to develop overall.  
Also since its more contained, I think it might have a higher chance of getting merged. It also has the nice side effect of developing a way to easily create checkpoints and save/reload them.

rocallahan, 2025-12-17:
I have merged a bunch of PRs to make RTLIL work safely when accessed read-only from multiple threads. Parallel writing is still not supported, but that's not an insurmountable problem since in many cases you can do analysis in parallel and queue up RTLIL edits to be applied by the main thread.