YosysHQ/yosys #5309: ECP5 LUT6 instantiation

peepo, 2025-08-21:
### Feature Description

Hi there,

Lattice Diamond enables LUT6 instantiation for ECP5 with something like:

LUT6 #(.init (64'h0001_0116_0116_1668))
I10 ( .A (buffer[7][0]), .B (buffer[7][1]), .C (buffer[0][7]), .D (buffer[0][1]), .E (buffer[1][7]), .F (buffer[1][0]), .Z (O10) );

Please could something similar be provided?

Thank you

My particular situation is that my Intel macmini died, I'd like a replacement, and without this, zugzwang....

Ravenslofty, 2025-08-22:
`assign O10 = 64'h0001_0116_0116_1668 >> {buffer[1][0], buffer[1][7], buffer[0][1], buffer[0][7], buffer[7][1], bufferp7][0]};`

closed 2025-08-22 by Ravenslofty

peepo, 2025-09-13:
Hi there, apologies for the delay.

On my setup, Yosys complains: 
ERROR syntax error, unexpected ','

This would be the first comma, in the example discussed: buffer[1][0],

Please advise further