YosysHQ/yosys #5318: pool::do_insert uses expiring value when calling do_hash

euphgh, 2025-09-01:
## Description:
In the first if branch of the [pool::do_insert](https://github.com/YosysHQ/yosys/blob/d4c4b2106697f62852cce03cd98eb78395a7779c/kernel/hashlib.h#L998) function, the rvalue is forwarded into entries.emplace_back. If it truly represents an rvalue, it becomes an expiring value after this line. However, the same value is then passed to do_hash, which may result in undefined or incorrect behavior.
```cpp
int do_insert(K &&rvalue, Hasher::hash_t &hash)
{
    if (hashtable.empty()) {
        entries.emplace_back(std::forward<K>(rvalue), -1);
        do_rehash();
        hash = do_hash(rvalue); // <-- potential issue: rvalue already consumed
    } else {
        ...
    }
    ...
}
```

## Proposed Fix:
A safer approach would be to make a copy of the key before forwarding the rvalue, and then use the copy when calling do_hash:
```cpp
auto key = rvalue;
entries.emplace_back(std::forward<K>(rvalue), -1);
do_rehash();
hash = do_hash(key);
```

## Notes:
- Currently, hash is not used, so this issue does not cause visible faults in the main codebase.
- However, this behavior surfaced as a bug while I was developing a plugin, which relies on this function.

I’d be happy to prepare a PR with the fix. Do you think this is worth addressing?

widlarizer, 2025-09-02:
I see. How did you detect or somehow expose the object expiration in your hash_into method?

euphgh, 2025-09-03:
I create a mini case [test_do_insert.cpp](https://github.com/user-attachments/files/22107929/test_do_insert.cpp), you can reproduce by command `yosys-config --build do_insert.so test_do_insert.cpp && yosys -m ./do_insert.so -p "test_do_insert"`

It use `Yosys::IdString` as an example. When the `index_` of IdString is 0, it indicates that this is an empty string or an expired string, because the move constructor sets the argument's `index_` to 0.
Therefore, I assert that the index is not 0 in the hash_into function. My expectation is that when I insert a non-empty IdString, this assert should not trigger, but actually the program triggered it.

[image]

closed 2025-09-09 by widlarizer