YosysHQ/yosys #5367: Issue with simlib.v, Unsupported: Verilog 1995 gate primitive: tran for Verilator

MrPoloGit, 2025-09-19:
### Version

0.57

### On which OS did this happen?

Linux

### Reproduction Steps

Issue from this commit
https://github.com/YosysHQ/yosys/commit/1251e92e3a988b33f2075118071f4454247041d6

1. Using Verilator 
```bash
Verilator 5.039 devel rev v5.038-8-gd89df33fc
```

2. Run folder of simlib.v
```bash
verilator --binary -f verilator.f --top-module simlib simlib.v
```

3. verilator.f
```bash

--timing
-j 0
-Wall
-Wno-fatal
--assert
--trace-fst
--trace-structs
--main-top-name "-"

--x-assign unique
--x-initial unique

-Werror-IMPLICIT
-Werror-USERERROR
-Werror-LATCH
```

### Expected Behavior

1. Parse all the Verilog files listed in verilator.f plus simlib.v.
2. Treat simlib as the top module.
3. Generate a C++ simulation model.
4. Link it into a runnable binary.
5. Able tp execute simulation.

### Actual Behavior

```bash
%Error-UNSUPPORTED: simlib.v:3229:6: Unsupported: Verilog 1995 gate primitive: tran
 3229 | tran connect[WIDTH-1:0] (A, B);
      |      ^~~~~~~
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=5.039
%Error: Exiting due to 2 error(s)
```

Ravenslofty, 2025-09-20:
@jix do you know of any other ways of expressing that primitive? Verilator doesn't like it.

jix, 2025-09-22:
@Ravenslofty my answer to that is essentially no. I do know of another way that should work in principle, directly connecting both ports to the same wire using ANSI port declarations with an explicit expression:

```
module \$connect #(parameter WIDTH = 0) (inout .A(W), inout .B(W));
	wire [WIDTH-1:0] W;
endmodule
```

This isn't useful, though as neither Yosys's read_verilog, Icarus Verilog nor Verilator currently support that (Verific does, confirming my interpretation that the spec allows this).

Given that, I think the best way forward is what @KrystalDelusion suggests, i.e. hiding it when `SIMLIB_NOCONNECT` is defined. Do we document those macros anywhere outside of `simlib.v`? 

---

For context, the `$connect` cell was added to represent bidirectional connections between inout ports (or connections with an ambiguous direction due to unknown port connections) while the design is in buffer normalized mode. This is done as buffer normalized mode doesn't use module level wire <-> wire connections, replacing those with primarily `$buf` buffers, but now also the `$connect` cell where a directional buffer can't be safely inserted. It will always be possible to get rid of it by running `clean` while the design isn't in buffer normalized mode.

So far, we don't support leaving the design buffer normalized outside of passes, so the only thing practically required is running `clean`. Therefore, hiding the `$connect` cell's definition from Verilator should be safe. If a user does end up with `$connect` cells they can always run `clean`. (Before use of bufnorm becomes more common, I also want to add a more targeted way of leaving bufnorm that only removes `$buf` and `$connect` without performing a full `clean`.)  

---

It's also unclear to me whether [Verilator's limited tri-state/inout support](https://verilator.org/guide/latest/languages.html#tri-inout) is sufficient to support the `$connect` cell at all, at least in the contexts where it would appear in practice, or whether the way it emulates it using multiple signals would require the connections to be inlined into the connecting module (which is what `clean` does). Lacking any supported syntax to directly connect multiple ports in a bidirectional way, I don't know how to test whether it could support this in principle.