YosysHQ/yosys #5401: `bufnorm` generates wrong code

sashimi-yzh, 2025-09-30:
### Version

Yosys 0.57+244 (git sha1 5fd2aecd9, clang++ 18.1.8 -fPIC -O3)

### On which OS did this happen?

Linux

### Reproduction Steps

The same verilog input from #5398 is used.

```verilog
// CacheMeta.sv
module tag_2x4(	// @[src/main/CacheMeta.scala:19:16]
  input        R0_addr,
               R0_en,
               R0_clk,
  output [3:0] R0_data,
  input        W0_addr,
               W0_en,
               W0_clk,
  input  [3:0] W0_data
);

  reg [3:0] Memory[0:1];	// @[src/main/CacheMeta.scala:19:16]
  always @(posedge W0_clk) begin	// @[src/main/CacheMeta.scala:19:16]
    if (W0_en & 1'h1)	// @[src/main/CacheMeta.scala:19:16]
      Memory[W0_addr] <= W0_data;	// @[src/main/CacheMeta.scala:19:16]
  end // always @(posedge)
  assign R0_data = Memory[R0_addr];	// @[src/main/CacheMeta.scala:19:16]
endmodule

module CacheMeta(	// @[src/main/CacheMeta.scala:6:7]
  input        clock,	// @[src/main/CacheMeta.scala:6:7]
               reset,	// @[src/main/CacheMeta.scala:6:7]
  output [3:0] io_tagOut,	// @[src/main/CacheMeta.scala:11:14]
  input  [3:0] io_tagIn,	// @[src/main/CacheMeta.scala:11:14]
  input        io_ridx,	// @[src/main/CacheMeta.scala:11:14]
               io_widx,	// @[src/main/CacheMeta.scala:11:14]
               io_wen // @[src/main/CacheMeta.scala:11:14]
);

  tag_2x4 tag_ext (	// @[src/main/CacheMeta.scala:19:16]
    .R0_addr (io_ridx),
    .R0_en   (1'b1),	// @[src/main/CacheMeta.scala:6:7]
    .R0_clk  (clock),
    .R0_data (io_tagOut),
    .W0_addr (io_widx),
    .W0_en   (io_wen),
    .W0_clk  (clock),
    .W0_data (io_tagIn)
  );	// @[src/main/CacheMeta.scala:19:16]
endmodule
```
```tcl
# yosys.tcl
yosys -import
read_verilog -sv CacheMeta.sv
synth -top CacheMeta
bufnorm
opt -purge
write_verilog -noattr -noexpr -nohex -nodec netlist.v
```

### Expected Behavior

Without the `bufnorm` command, the netlist is correct.

```verilog
/* Generated by Yosys 0.57+244 (git sha1 5fd2aecd9, clang++ 18.1.8 -fPIC -O3) */

module CacheMeta(clock, reset, io_tagOut, io_tagIn, io_ridx, io_widx, io_wen);
  input clock;
  wire clock;
  input reset;
  wire reset;
  output [3:0] io_tagOut;
  wire [3:0] io_tagOut;
  input [3:0] io_tagIn;
  wire [3:0] io_tagIn;
  input io_ridx;
  wire io_ridx;
  input io_widx;
  wire io_widx;
  input io_wen;
  wire io_wen;
  tag_2x4 tag_ext (
    .R0_addr(io_ridx),
    .R0_clk(clock),
    .R0_data(io_tagOut),
    .R0_en(1'b1),
    .W0_addr(io_widx),
    .W0_clk(clock),
    .W0_data(io_tagIn),
    .W0_en(io_wen)
  );
endmodule

module tag_2x4(R0_addr, R0_en, R0_clk, R0_data, W0_addr, W0_en, W0_clk, W0_data);
  input R0_addr;
  wire R0_addr;
  input R0_en;
  wire R0_en;
  input R0_clk;
  wire R0_clk;
  output [3:0] R0_data;
  wire [3:0] R0_data;
  input W0_addr;
  wire W0_addr;
  input W0_en;
  wire W0_en;
  input W0_clk;
  wire W0_clk;
  input [3:0] W0_data;
  wire [3:0] W0_data;
  wire _00_;
  wire _01_;
  wire [3:0] \Memory[0] ;
  wire [3:0] \Memory[1] ;
  \$_AND_  _02_ (
    .A(W0_en),
    .B(W0_addr),
    .Y(_01_)
  );
  \$_ANDNOT_  _03_ (
    .A(W0_en),
    .B(W0_addr),
    .Y(_00_)
  );
  \$_MUX_  _04_ (
    .A(\Memory[0] [0]),
    .B(\Memory[1] [0]),
    .S(R0_addr),
    .Y(R0_data[0])
  );
  \$_MUX_  _05_ (
    .A(\Memory[0] [1]),
    .B(\Memory[1] [1]),
    .S(R0_addr),
    .Y(R0_data[1])
  );
  \$_MUX_  _06_ (
    .A(\Memory[0] [2]),
    .B(\Memory[1] [2]),
    .S(R0_addr),
    .Y(R0_data[2])
  );
  \$_MUX_  _07_ (
    .A(\Memory[0] [3]),
    .B(\Memory[1] [3]),
    .S(R0_addr),
    .Y(R0_data[3])
  );
  \$_DFFE_PP_  \Memory_reg[1][0]  /* _08_ */ (
    .C(W0_clk),
    .D(W0_data[0]),
    .E(_01_),
    .Q(\Memory[1] [0])
  );
  \$_DFFE_PP_  \Memory_reg[1][1]  /* _09_ */ (
    .C(W0_clk),
    .D(W0_data[1]),
    .E(_01_),
    .Q(\Memory[1] [1])
  );
  \$_DFFE_PP_  \Memory_reg[1][2]  /* _10_ */ (
    .C(W0_clk),
    .D(W0_data[2]),
    .E(_01_),
    .Q(\Memory[1] [2])
  );
  \$_DFFE_PP_  \Memory_reg[1][3]  /* _11_ */ (
    .C(W0_clk),
    .D(W0_data[3]),
    .E(_01_),
    .Q(\Memory[1] [3])
  );
  \$_DFFE_PP_  \Memory_reg[0][0]  /* _12_ */ (
    .C(W0_clk),
    .D(W0_data[0]),
    .E(_00_),
    .Q(\Memory[0] [0])
  );
  \$_DFFE_PP_  \Memory_reg[0][1]  /* _13_ */ (
    .C(W0_clk),
    .D(W0_data[1]),
    .E(_00_),
    .Q(\Memory[0] [1])
  );
  \$_DFFE_PP_  \Memory_reg[0][2]  /* _14_ */ (
    .C(W0_clk),
    .D(W0_data[2]),
    .E(_00_),
    .Q(\Memory[0] [2])
  );
  \$_DFFE_PP_  \Memory_reg[0][3]  /* _15_ */ (
    .C(W0_clk),
    .D(W0_data[3]),
    .E(_00_),
    .Q(\Memory[0] [3])
  );
endmodule
```

### Actual Behavior

With the `bufnorm` command, a lot of signals in `CacheMeta` are connected with `clock`. There are even assignments to input ports.

```verilog
/* Generated by Yosys 0.57+244 (git sha1 5fd2aecd9, clang++ 18.1.8 -fPIC -O3) */

module CacheMeta(clock, reset, io_tagOut, io_tagIn, io_ridx, io_widx, io_wen);
  input clock;
  wire clock;
  input reset;
  wire reset;
  output [3:0] io_tagOut;
  wire [3:0] io_tagOut;
  input [3:0] io_tagIn;
  wire [3:0] io_tagIn;
  input io_ridx;
  wire io_ridx;
  input io_widx;
  wire io_widx;
  input io_wen;
  wire io_wen;
  tag_2x4 tag_ext (
    .R0_addr(clock),      // BUG: incorrectly connect to clock
    .R0_clk(clock),
    .R0_data(io_tagOut),
    .R0_en(1'b1),
    .W0_addr(clock),  // BUG: incorrectly connect to clock
    .W0_clk(clock),
    .W0_data(io_tagIn),
    .W0_en(clock)   // BUG: incorrectly connect to clock
  );
  assign io_ridx = clock;   // BUG: assign to an input port
  assign io_wen = clock;   // BUG: assign to an input port
  assign io_widx = clock;   // BUG: assign to an input port
  assign reset = clock;   // BUG: assign to an input port
endmodule

module tag_2x4(R0_addr, R0_en, R0_clk, R0_data, W0_addr, W0_en, W0_clk, W0_data);
  input R0_addr;
  wire R0_addr;
  input R0_en;
  wire R0_en;
  input R0_clk;
  wire R0_clk;
  output [3:0] R0_data;
  wire [3:0] R0_data;
  input W0_addr;
  wire W0_addr;
  input W0_en;
  wire W0_en;
  input W0_clk;
  wire W0_clk;
  input [3:0] W0_data;
  wire [3:0] W0_data;
  wire _00_;
  wire [3:0] \Memory[0] ;
  wire [3:0] \Memory[1] ;
  \$_ANDNOT_  _01_ (
    .A(R0_addr),
    .B(R0_addr),
    .Y(_00_)
  );
  \$_MUX_  _02_ (
    .A(\Memory[0] [0]),
    .B(\Memory[1] [0]),
    .S(R0_addr),
    .Y(R0_data[0])
  );
  \$_MUX_  _03_ (
    .A(\Memory[0] [1]),
    .B(\Memory[1] [1]),
    .S(R0_addr),
    .Y(R0_data[1])
  );
  \$_MUX_  _04_ (
    .A(\Memory[0] [2]),
    .B(\Memory[1] [2]),
    .S(R0_addr),
    .Y(R0_data[2])
  );
  \$_MUX_  _05_ (
    .A(\Memory[0] [3]),
    .B(\Memory[1] [3]),
    .S(R0_addr),
    .Y(R0_data[3])
  );
  \$_DFFE_PP_  \Memory_reg[1][0]  /* _06_ */ (
    .C(R0_addr),
    .D(W0_data[0]),
    .E(R0_addr),
    .Q(\Memory[1] [0])
  );
  \$_DFFE_PP_  \Memory_reg[1][1]  /* _07_ */ (
    .C(R0_addr),
    .D(W0_data[1]),
    .E(R0_addr),
    .Q(\Memory[1] [1])
  );
  \$_DFFE_PP_  \Memory_reg[1][2]  /* _08_ */ (
    .C(R0_addr),
    .D(W0_data[2]),
    .E(R0_addr),
    .Q(\Memory[1] [2])
  );
  \$_DFFE_PP_  \Memory_reg[1][3]  /* _09_ */ (
    .C(R0_addr),
    .D(W0_data[3]),
    .E(R0_addr),
    .Q(\Memory[1] [3])
  );
  \$_DFFE_PP_  \Memory_reg[0][0]  /* _10_ */ (
    .C(R0_addr),
    .D(W0_data[0]),
    .E(_00_),
    .Q(\Memory[0] [0])
  );
  \$_DFFE_PP_  \Memory_reg[0][1]  /* _11_ */ (
    .C(R0_addr),
    .D(W0_data[1]),
    .E(_00_),
    .Q(\Memory[0] [1])
  );
  \$_DFFE_PP_  \Memory_reg[0][2]  /* _12_ */ (
    .C(R0_addr),
    .D(W0_data[2]),
    .E(_00_),
    .Q(\Memory[0] [2])
  );
  \$_DFFE_PP_  \Memory_reg[0][3]  /* _13_ */ (
    .C(R0_addr),
    .D(W0_data[3]),
    .E(_00_),
    .Q(\Memory[0] [3])
  );
  assign R0_clk = R0_addr;       // BUG: assign to an input port
  assign R0_en = R0_addr;       // BUG: assign to an input port
  assign W0_addr = R0_addr;  // BUG: assign to an input port
  assign W0_clk = R0_addr;     // BUG: assign to an input port
  assign W0_en = R0_addr;     // BUG: assign to an input port
endmodule
```

jix, 2025-09-30:
The `bufnorm` command is experimental (and marked as such) and so far only works in specific scenarios. In particular we do not support entering bufnorm mode followed by running arbitrary passes like `opt`. Can you tell us more about what you are trying to do with `bufnorm`?

sashimi-yzh, 2025-09-30:
Thank you for the explanation about the `bufnorm` command. I am trying to use it to optimize my processor. I will try this command again after it is stable.